mudpi/mudpi-core #5: Threading locks up during KeyboardInterrupt shutdown

trial123Zel, 2020-05-31:
##  **Issue**
-----
MudPi gets stuck during shutdown due to threading issue. To fully exit mudpi.py, you must KeyboardInterrupt 2 more times. This seems to have started after integrating more sensors and nodes (aka it used to successfully shut down with less integrated pieces)

## **MudPi Version**
------
v0.8.10

## **MudPi Config**
-----
Initializing Garden Control...           Complete
Preparing Threads for Workers...         Complete
No Camera Found to Load
Humidity Sensor (Pi) 18...               Ready
Loading Pi Sensor Worker...
Pi Sensor Worker [4 Sensors]...          Running
Float Sensor (Pi) 17...          Ready
Float Sensor (Pi) 27...          Ready
Loading Pi Sensor Worker...
Pi Sensor Worker [4 Sensors]...          Running
Relay Worker pump1...                    Ready
Relay Worker pump1...                    Running
Relay Worker water_ch1...                        Ready
Relay Worker water_ch1...                        Running
Relay Worker water_ch2...                        Ready
Relay Worker water_ch2...                        Running
Relay Worker water_ch3...                        Ready
Relay Worker water_ch3...                        Running
Relay Worker water_ch4...                        Ready
Relay Worker water_ch4...                        Running
Relay Worker water_ch5...                        Ready
Relay Worker water_ch5...                        Running
Relay Worker water_ch6...                        Ready
Relay Worker water_ch6...                        Running
Relay Worker water_ch7...                        Ready
Relay Worker water_ch7...                        Running
Relay Worker water_ch8...                        Ready
Relay Worker water_ch8...                        Running
slave2 -> Wifi Connection                Success
Soil Sensor 5...                         Ready
slave2 Node Worker [S: 1][C: 0]...       Running
Loading Triggers...
Trigger Worker [1 Triggers]...           Running
MudPi Server...                          Running
MudPi Garden Control...                  Online

## **Python Errors**
-----
Switch Relay pump1 state to 0
Pi Temp: {'temperature': 69.8, 'humidity': 41.0}
{'soil1': 1023}
{'soil1': 1023}
{'soil1': 1023}
{'soil1': 1023}
Switch Relay pump1 state to 0
^CMudPi Shutting Down...                                <---KeyboardInterrupt (CTRL + C) given
Server Shutdown...                       Complete
slave2 Controls Shutting Down...         Complete
Relay Worker water_ch4 Shutting Down...  Complete
Relay Worker water_ch5 Shutting Down...  Complete
Relay Worker water_ch8 Shutting Down...  Complete
Relay Worker water_ch1 Shutting Down...  Complete
Relay Worker pump1 Shutting Down...      Complete
Relay Worker water_ch6 Shutting Down...  Complete
Relay Worker water_ch3 Shutting Down...  Complete
Relay Worker water_ch2 Shutting Down...  Complete
Relay Worker water_ch7 Shutting Down...  Complete
slave2 Sensors Shutting Down...  Complete
slave2 Shutting Down...          Complete
Trigger Worker Shutting Down...          Complete
Pi Sensor Worker Shutting Down...        Complete
^CTraceback (most recent call last):
  File "mudpi.py", line 259, in <module>
    thread.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
^CException ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
Exception ignored in: <function Connection.__del__ at 0x76534390>
Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/redis/connection.py", line 537, in __del__
  File "/home/pi/.local/lib/python3.7/site-packages/redis/connection.py", line 667, in disconnect
TypeError: catching classes that do not inherit from BaseException is not allowed

olixr, 2020-06-01:
You should end the program and wait for a few minutes to be sure it actually hangs. All the thread wait once the program ends to join and shutdown. 

If you have nodes, especially wireless ones, they often only communicate every 30secs if not more depending on your intervals. The thread for those nodes waits until the next command to issue the shutdown. You are likely just waiting on some of your nodes to respond.

Try shutting down with and without nodes to get a feel of the time needed.

trial123Zel, 2020-06-01:
Thanks for the explanation. I'll close this issue.

closed 2020-06-01 by trial123Zel