mudpi/mudpi-core #8: Arduinio is not reconized

jab375, 2020-06-27:
I followed the Arduino guide and Mudpi seems to have an issue connecting to the slave Arduino UNO i have. Right now i have nanpi firmware on my UNO and installed nanpi to the pi according to the link below. After doing this I notice a /dev/ttyACM0  comes and go when the Uno USB port is plugged/unplugged into the PI USB port. I assume ACMO is the right port and entered that into the mudpi config file that was generated by mudpi.app website. Not sure what I'm doing wrong. Maybe a bug?

https://mudpi.app/guides/9
https://nanpy.github.io/

Boot screen of Mudpi shows successful serial for my UNO and right under it complains about the Node connection.
` __  __           _ _____ _
|  \/  |         | |  __ (_)
| \  / |_   _  __| | |__) |
| |\/| | | | |/ _` |  ___/ |
| |  | | |_| | (_| | |   | |
|_|  |_|\__,_|\__,_|_|   |_|
_________________________________________________

Eric Davisson @theDavisson
Version:  0.8

Initializing Garden Control...           Complete
Preparing Threads for Workers...         Complete
No Camera Found to Load
Loading Pi Sensor Worker...
Pi Sensor Worker [2 Sensors]...          Running
[Arduino] Serial Connection      Success
Invalid or no Nodes found to Load
No Actions Found to Load
No Triggers Found to Load
MudPi Server...                          Running
MudPi Garden Control...                  Online
_________________________________________________
`

jab375, 2020-06-27:
I'm able to blink the led on pin 13 with this website setup. I think Mudpi has a bug somewhere

https://www.raspberrypi.org/forums/viewtopic.php?t=46881&start=50

olixr, 2020-06-28:
This is because you should be using the address of your device, not the examples. In the 'address' section you should be placing your connection to the slave. You say that you connected on '/dev/ttyACM0', then this should be the address. Everything is working correct it just seems you need to update you configurations. 

Please have a read through the documentation https://mudpi.app/docs/ to understand the core workings more. This will help when reading the guides to know what you should be updating. 

Happy Growing!

jab375, 2020-06-28:
I have updated my device connection and it throws the invalid or not connected error after I reboot the service and uno. I used the Mudpi.app config generator to the output below.

`{
    "name": "MudPi",
    "version": 0.8,
    "debug": false,
    "redis": {
        "host": "127.0.0.1",
        "port": 6739
    },
    "relays": [],
    "workers": [
        {
            "type": "sensor",
            "sensors": []
        }
    ],
    "server": {
        "host": "127.0.0.1",
        "port": 6602
    },
    "nodes": [
        {
            "name": "New Garden Area",
            "address": "/dev/ttyACM0",
            "sensors": [
                {
                    "pin": "0",
                    "is_digital": true,
                    "type": "Humidity",
                    "name": "Humidity0"
                },
                {
                    "pin": 1,
                    "is_digital": true,
                    "type": "Humidity",
                    "name": "Humidity1"
                },
                {
                    "pin": "2",
                    "is_digital": false,
                    "type": "Humidity",
                    "name": "Humidity2"
                }
            ]
        }
    ]
}
`

olixr, 2020-06-28:
Please format the code into code blocks in the future, it is very hard to read otherwise. Try stripping things down and testing your configuration on a simple level. Below is a config I updated formatting quickly. The issue is very likely formatting in your configuration. Comma placements and spacing is important.  

```
{ 
	"name": "MudPi", 
	"version": 0.8, 
	"debug": false, 
	"server": { 
		"host": "127.0.0.1", 
		"port": 6602 
	}, 
	"redis": { 
		"host": "127.0.0.1", 
		"port": 6739 
	}, 
	"relays": [], 
	"workers": [], 
	"nodes": [ 
		{ 
			"name": "New Garden Area", 
			"address": "/dev/ttyACM0", 
			"sensors": [ 
				{ 
					"pin": 0,
					"type": "Humidity", 
					"name": "Humidity0" 
				}
			],
			"controls":[]
		} 
	] 
} 
```

I would also highly recommend joining the discord to ask folks for some help too https://discord.gg/daWg2YH. The community has been awesome in getting everyone's builds up and running.

closed 2020-07-18 by olixr