mudpi/mudpi-core #9: DHT22 and Ardunio Issue

mexicanhaze, 2020-07-05:
When a DHT22 sensor is connected on an Arduino it gives out false readings (with a temperature usually at 0°C)
Putting ```model='22'``` in the __init__ and ```if len(self.type) == 2``` in init_sensor solves the issue but probably prevents other sensor types from working , I don't know as I don't have any others to test
https://cdn.discordapp.com/attachments/614176724938129408/729429116201467924/unknown.png

olixr, 2020-08-11:
I have tested a DHT11 and DHT22 setup on the Pi 4 and Pi zero. I ran this using MudPi V0.8.11 and was able to get readings from both. I noted that if you put in a integer for the config it would not properly cast that to a string thus causing a default to the model 11. 

I pushed an update coming in v0.8.12 that will cast int to string for the model and would help as a way to avoid this error. Hope this helps. I will close this when I merge the updates to master.

olixr, 2020-08-15:
I have had a working setup now for a few days without issue. I suggest restarting the device and power cycling the component if the error pops up again. Also as of PR https://github.com/mudpi/mudpi-core/pull/11 which brings MudPi to v0.9.0 there was a bunch of refactors. If there is still an issue we can revisit and reopen.

closed 2020-08-15 by olixr