mudpi/mudpi-core #20: Improve time scheduling

SecT0uch, 2020-11-19:
Hey,

I'm thinking about a way to improve the Time Trigger scheduling.

My initial concern was to be able to detect if we missed a schedule and correct the relay position. 
Exemple:
* I have 2 daily triggers to switch the lights on and off, let's say: On at 8:00, Off at 21:00
* If if I start mudpi for the first time at 10:00, I want my lights to be "On" because we are in schedule

I know this is possible use pycron's `has_been()` function.

Shouldn't we change the Time Trigger declaration or create a new trigger type `daily_schedule`?
It would take 2 options, let's say `schedule_on` and `schedule_off`.

In the init we would add something like:
```python
from datetime import date
from datetime import datetime

today = date.today()
midnight = datetime.combine(today, datetime.min.time())

if pycron.has_been(schedule_on, midnight) and not pycron.has_been(schedule_off, midnight):   # if in schedule
   <Turn the relay On>
```

Do you have any better idea on how to implement this?

I can contribute to implement that if needed.

SecT0uch, 2020-12-10:
What if I set 2 schedules linked to 2 events as such:
* `* 8-20 * * *`: trigger an event `Switch`  1 every minute from 8:00 to 20:59
*  `* 0-7,21-23 * * *`: trigger an event `Switch`  0 every minute from 0:00 to 7:59 and 21:00 to 23:59

Would it be a problem to trigger the same event every minute ?
I see there are checks in `turn_on()` and `turn_off()` functions, but what about Redis spamming ?

olixr, 2021-03-12:
These are wonderful suggestions. Check out the new v0.10.0 that was in the works.  https://github.com/mudpi/mudpi-core/commit/809f998aeec62af7ddef827326065177e5bf7c50

I think some of the issue you had with triggers will be resolved. However I do have plans for more trigger options following the update. One of them is very similar to the suggestion you make. Let me know if the new changes get the behavor you are looking for. If not perhaps the triggers I have planned are what you are thinking of then. I want to provide the option to use cron strings and also just time based (timers) to give a variety of choices. 

Let me know what you think of the updates and if this resolves the issues.

SecT0uch, 2021-03-12:
My board is currently used in production, but can wait to put my hands on the new version.
I will let you know as soon as I'm able to test :)

closed 2021-03-12 by SecT0uch