mudpi/mudpi-core #21: Hamburger Menu not clickable on mobile

BillyHilly, 2020-11-23:
Hello,
First off, great work on this, it's awesome!
I too am a web developer, and I've thoroughly enjoyed getting to know both Raspberry Pi and MudPi. You did alot of great work under the hood. Very impressive.
I hope to be able to contribute to the project in the future.
Until then, just wanted to alert you to the an issue I noticed.
When I visit my localhost/ site MudPi sets up, on my iPhone 6s the hamburger menu doesn't open / not clickable.
I didn't look into it yet, but my guess would be it just needs a position and higher z-index set via CSS. It's like it's behind the main layer and thus unclickable. 
Anyways, wanted to let you know and also say awesome job!

yeyeto2788, 2020-11-23:
Hey @BillyHilly,

I think this issue should be opened on the [assistant](https://github.com/mudpi/assistant) repository. But let's wait for @olixr decision on this one.

Would be also helpful if you add more details on how to reproduce the error. Things like:
* Browser used.
* OS (iOS version)
* MudPi version used

BillyHilly, 2020-11-23:
Thanks @yeyeto2788 
Not sure on the Assistant repo since the server/website is part of MudPi. But whichever is most applicable is fine by me — whatever's decided.
* Browser: Safari on iOS v.14.2
* OS: iOS v.14.2
* MudPi v.0.8
 
I just checked the issue directly on the rPi browser and desktop Chrome/Safari/FF, shrunk the window to mobile size and the hamburger nav shows.
Same issue across the board.
I noticed a `@click="toggleNav" attribute set on the Hamburger SVG, but it doesn't seem to be working or throwing an error. Not sure I can be of much further help since there's no error and it's JS based (I always use CSS only method for hamburgers).
Hope that helps!

SecT0uch, 2020-12-09:
Same issue here on Firefox Android.
The website seems to be in the ui repo

olixr, 2021-03-12:
Yes the UI is under another repo. Closing this issue and we can open it under there. 

The menu though is not backed by any javascript action. Its just cosmetic at the moment. The UI is getting a replaced with my internal one when its ready to release. If you want to try and update the UI menu to work though that would be awesome! Otherwise I don't know when I will get to it in favor of the new dashboard in the pipeline.

closed 2021-03-12 by olixr