mudpi/mudpi-core #23: Fails to run on Rasp Pi A

tezcooling, 2020-12-17:
I've been trying to set MudPi up on an old Raspberry Pi A.

I've managed to get the installer to complete without any errors. It's running buster fine. I get the setup when I log in via the web browser setup but I soon as I click "Get Started" I get a "500 internal server error"

If I try to run MudPi Manually from the CLI I get the following error:

Traceback (most recent call last):
  File "mudpi.py", line 27, in <module>
    from workers.adc_worker import ADCMCP3008Worker
  File "/etc/mudpi/core/workers/adc_worker.py", line 14, in <module>
    class ADCMCP3008Worker:
  File "/etc/mudpi/core/workers/adc_worker.py", line 23, in ADCMCP3008Worker
    '5': board.D5,
AttributeError: module 'board' has no attribute 'D5'

I've been googling all afternoon but can't seem to narrow down the problem?

olixr, 2021-03-12:
The specific error you are getting is coming from Adafruit blinka. Perhaps blinka does not support that model pi?

The other thing is I only have tested and provide support for the Pi 2B+ and older. There were some GPIO/ firmware changes prior to the 2B that makes it harder to offer all the features too. 

Try taking a look at adafruit blinka and if it supports that model Pi.

closed 2021-03-12 by olixr

yeyeto2788, 2021-03-15:
From what I can see the `D5` pin is not declared on the board definition on the `blinka` library. It get it's definition from here:

https://github.com/adafruit/Adafruit_Blinka/blob/master/src/board.py#L56 which is defined on this file https://github.com/adafruit/Adafruit_Blinka/blob/master/src/adafruit_blinka/board/raspberrypi/raspi_1b_rev2.py so I guess it would be great to make the class `ADCMCP3008Worker` on mudpi-core a much more customizable object.