mudpi/mudpi-core #24: RPI 4 error - ImportError: cannot import name 'Beaglebone_Black_Driver' from 'Adafruit_DHT'

icyspace, 2021-01-31:
The following error message occurs after clean install on RPI when using this configuration. 

...
{
    "name": "MudPi",
    "debug": false,
    "logging": {
        "file_log_level": "debug",
        "terminal_log_level": "debug",
        "file": "mudpi.log"
    },
    "workers":[{
        "type":"sensor",
        "sensors": [
            {
                "type":"Humidity",
                "pin": 24,
                "key":"weather_sensor"
            }
        ]
    }]
}
...

```
File "/usr/local/lib/python3.7/dist-packages/Adafruit_DHT/Beaglebone_Black.py", line 24, in <module>
    from . import Beaglebone_Black_Driver as driver
ImportError: cannot import name 'Beaglebone_Black_Driver' from 'Adafruit_DHT' (/usr/local/lib/python3.7/dist-packages/Adafruit_DHT/init.py)
```

Based on the research this is a common problem with the old Adafruit_DHT on RPI4.  the suggested workaround here resolved my issue. 

https://stackoverflow.com/questions/63232072/cannot-import-name-beaglebone-black-driver-from-adafruit-dht

tpauwels, 2021-03-03:
Thanks! Had the same issue. Solved with this workaround.

olixr, 2021-03-12:
Can you test this on the new v0.10.0?

v0.10.0: https://github.com/mudpi/mudpi-core/commit/809f998aeec62af7ddef827326065177e5bf7c50

Seems like this is an issue with the Adafruit_DHT that we should raise the issue there.

icyspace, 2021-03-13:
I will test and report back.

The comments in the stack overflow indicated the library is no longer supported. 

From stackoverflow
"The real problem is that AdaFruit no longer supports nor updates this old Adafruit_DHT library. The new library from AdaFruit for this sensor is "Adafruit_CircuitPython_DHT" can be found here. It could be a good idea to update your code to this new library."

yeyeto2788, 2021-03-15:
Hey @icyspace, 

You're completely right, that is why everything was change to use Adafruit's Blinka library so we avoid this kind of issues but also to be compatible with more boards.

@olixr from a quick look at the version proposed the `mudpi/extensions/dht/sensor.py` seems like it is still using the old library and should be replace by this one https://github.com/adafruit/Adafruit_CircuitPython_DHT. That shouldn't be much of a hassle.
If you make the changes I can also test this one on the OPI lite

olixr, 2021-03-15:
I reviewed the `dht` extension changes and I thought that I have already made this change. Have a look at the `extension.json` file which lists`"requirements": ["adafruit-blinka", "adafruit-circuitpython-dht"]`.

Also in the `sensor.py` I have the import listed here https://github.com/mudpi/mudpi-core/blob/809f998aeec62af7ddef827326065177e5bf7c50/mudpi/extensions/dht/sensor.py#L8 

Have a look through and confirm that. The old library is no longer used.

yeyeto2788, 2021-03-18:
@icyspace, @olixr 

We're on a good path, I finally got the sensor to work:

```console
███╗   ███╗██╗   ██╗██████╗ ██████╗ ██╗
████╗ ████║██║   ██║██╔══██╗██╔══██╗██║
██╔████╔██║██║   ██║██║  ██║██████╔╝██║
██║╚██╔╝██║██║   ██║██║  ██║██╔═══╝ ██║
██║ ╚═╝ ██║╚██████╔╝██████╔╝██║     ██║
╚═╝     ╚═╝ ╚═════╝ ╚═════╝ ╚═╝     ╚═╝
____________________________________________________________

Eric Davisson @MudPiApp
https://mudpi.app
Version:  0.10.0

Loading MudPi Configs ............................... Complete
DEBUG MODE ENABLED
Config path: /home/yeyeto2788/workspace/mudpi-core/mudpi.json
Current Directory: /home/yeyeto2788/workspace/mudpi-core
Initializing Logger  ................................ Complete
Preparing State Manager  ............................ Complete
Preparing Event System for redis  ................... Pending
Event System Ready on redis   ....................... Connected
Initializing Core  .................................. Complete
_________________ Detecting Configurations _________________
Detecting Configurations ............................ Pending
Detected 1 Non-Core Configurations .................. Complete
Preparing 1 Configurations to be Loaded  ............ Pending
Sensor Ready for Import ............................. Ready
1 Configurations Ready to Load  ..................... Complete
_____________________ Load Extensions ______________________
Loading 1 Configurations into Extensions  ........... Pending
Loading: sensor
Importing Sensor .................................... Pending
Imported Sensor ..................................... Success
Dht Ready for Import ................................ Ready
Dht requirements .................................... Installing
Installing package adafruit-circuitpython-dht==3.5.5
Importing dht:sensor ................................ Pending
Imported dht:sensor ................................. Success
Configuration for dht:sensor ........................ Validated
Example Ready for Import ............................ Ready
Importing example:sensor ............................ Pending
Imported example:sensor ............................. Success
Configuration for example:sensor .................... Validated
Sensor Configuration ................................ Validated
Initializing Sensor ................................. Pending
Importing Dht ....................................... Pending
Imported Dht ........................................ Success
Dht Configuration ................................... Validated
Extension dht did not define an `init()` method.
Initializing Dht .................................... Pending
Initialized Dht ..................................... Success
Sensor Initializing: DHT
Importing Example ................................... Pending
Imported Example .................................... Success
Example Configuration ............................... Validated
Extension example did not define an `init()` method.
Initializing Example ................................ Pending
Initialized Example ................................. Success
Initialized Sensor .................................. Success
Loaded 3 Extensions  ................................ Complete
MudPi Fully Loaded .................................. Complete
______________________ Start Systems _______________________
Starting All Workers  ............................... Pending
Worker sensor.dht.30  ............................... Starting
Worker sensor.dht.30  ............................... Started
Worker sensor.example.30  ........................... Starting
[... 87 lines cut ...]
State Changed: example_sensor_1 - 7 @ 2021-03-18 08:57:38
DHT sensor not found, check wiring
DHT Reading was Invalid. Trying again next cycle.
State Changed: example_sensor_1 - 2 @ 2021-03-18 08:58:08
State Changed: example_sensor_1 - 6 @ 2021-03-18 08:58:38
DHT sensor not found, check wiring
DHT Reading was Invalid. Trying again next cycle.
State Changed: example_sensor_1 - 5 @ 2021-03-18 08:59:08
Checksum did not validate. Try again.
DHT Reading was Invalid. Trying again next cycle.
State Changed: example_sensor_1 - 4 @ 2021-03-18 08:59:38
DHT sensor not found, check wiring
DHT Reading was Invalid. Trying again next cycle.
State Changed: example_sensor_1 - 2 @ 2021-03-18 09:00:08
State Changed: example_sensor_1 - 1 @ 2021-03-18 09:00:38
^C____________________________________________________________
Stopping All Workers for Shutdown   ................. Pending
Worker sensor.dht.30  ............................... Stopping
Worker sensor.dht.30  ............................... Offline
Worker sensor.example.30  ........................... Stopping
Worker sensor.example.30  ........................... Offline
All MudPi Systems   ................................. Offline
```

There are a couple of things to be done in order to be able to properly read (Pin validation on mudpi) the sensor but my guess is that also my sensor is not working as expected as you could see in `https://github.com/adafruit/Adafruit_CircuitPython_DHT/issues/66`

olixr, 2021-04-16:
Doing some cleanup. Where are we at with this issue? I included the `dht_legacy` extension which uses the old library when the new blinka library causes issues. I have been able to get the new library working on a Pi4 with both DHT11 and DHT22. However on the Pi zero (ARMv6) I returned to the old library through dht_legacy for better read results. 

If this solves the issues until blinka fixes their DHT timing issue then we can close this. Otherwise let me know if there is more to address.

icyspace, 2021-04-18:
@olixr  I don't think there is much more that can be done.  You support both the new and old library.  I was able to get reading on both.  For me the legacy seemed much more reliable on rpi4.

olixr, 2021-04-18:
Ok thanks @icyspace  I will close this out for now. DHT sensors are known for being a bit tricky as well with reliable readings.

closed 2021-04-18 by olixr