mudpi/mudpi-core #25: Initial configuration error

drmaddy736, 2021-03-13:
I have installed mudpi, mudpi assistant and UI , without error. I could connect to AP(192.168.2.1). But then after clicking  'getting started' next page is not loading. Also in terminal, running mudpi.py gives error as 
mudpi.config file not found

olixr, 2021-03-15:
V0.10.0 is in the `feature` branch. This is a bug update to the system and will be suggested to update in the coming days. 

Your errors sound like you are missing a config file. You should create your own before running MudPi. An example is provided for you to start from but you still need to either rename it or make a copy. i.e. `cp mudpi.config.example mudpi.config` 

In regards to the dashboard that issue should be manged on the other repo. Its likely the error because of the updates to the core and the dashboard is still not fully updated to reflect them yet.

drmaddy736, 2021-03-15:
> V0.10.0 is in the feature branch. This is a big update to the system and

Yes... initial error was due to missing config file but then I created on
but very next error was no redis module then I copy pasted redis folder
into mudpi/core but then next error on running mudpi.py was ;modulenotfound
error , no module named 'board'. So I think mudpi.py script is not able to
get Access to other installed modules. I did try to find folder where all
modules would be there but I couldn't. By the way I am physician here India
with love for farming and little bit of tinkering with softwares.

Thank for Replay
Namaste 🙏

olixr, 2021-04-16:
Hey @drmaddy736 did you resolve you configuration issue?

drmaddy736, 2021-04-16:
No not yet.

olixr, 2021-04-18:
If you need more help please provide copies of your log files, configs and screenshots of any errors. Otherwise it can be hard to figure out the root of your issue. 

At initial notes it seems that you are missing a config file and just need to create one. 

If you need info on the config format you can read more here: https://mudpi.app/docs/configuration-basics

drmaddy736, 2021-04-18:
Will do, thanks!

olixr, 2021-05-25:
The v0.10.0 has been merged to the master branch. I am closing this issue as the config should be resolved with the update.

closed 2021-05-25 by olixr