mudpi/mudpi-core #29: support for I2C relays

duffrecords, 2021-03-30:
I have a [GeeekPi relay expansion board](https://www.amazon.com/gp/product/B07Q2P9D7K) mounted on my Raspberry Pi.  It communicates over I2C and, although I can toggle each relay from the command line, it looks like there's only support for GPIO-controlled relays at the moment.  I'd be happy to contribute an extension for this but I'm still getting familiarized with the MudPi platform.  Can you give me a basic idea of the scope of such a feature?  I get the sense that you're doing some non-trivial refactoring with this project and I'd want to stay consistent with that.

olixr, 2021-03-30:
Hey @duffrecords I am happy to have you using MudPi. You can get support for I2C relays by adding in support through an extension. Here are some docs I typed on the extension system https://mudpi.app/docs/extension-basics.

You will probably use the already existing `i2c` extension an just be adding a toggle through a`toggle.py` [interface.](https://mudpi.app/docs/developers-interfaces) A good place to reference is the `example` extension which has a bunch of components and interface just for testing. It can help show some of the structure. 

Let me know any questions as you go and I would be glad to help review updates.

duffrecords, 2021-04-08:
I have what I think should be a working toggle interface for the `i2c` extension, although I'm not able to test it yet.  I installed the feature branch (since I don't see any reference to extensions in master or any tags) but the server won't start because of this error:
```
Traceback (most recent call last):
  File "/usr/local/bin/mudpi", line 11, in <module>
    load_entry_point('mudpi==0.10.0', 'console_scripts', 'mudpi')()
  File "/usr/local/lib/python3.7/dist-packages/mudpi-0.10.0-py3.7.egg/mudpi/__main__.py", line 92, in main
    manager.load_mudpi_core()
  File "/usr/local/lib/python3.7/dist-packages/mudpi-0.10.0-py3.7.egg/mudpi/managers/core_manager.py", line 36, in load_mudpi_core
    self.mudpi.load_core()
  File "/usr/local/lib/python3.7/dist-packages/mudpi-0.10.0-py3.7.egg/mudpi/core.py", line 128, in load_core
    self.states = StateManager(self, self.config.get('mudpi', {}).get('events', {}).get('redis'))
  File "/usr/local/lib/python3.7/dist-packages/mudpi-0.10.0-py3.7.egg/mudpi/managers/state_manager.py", line 36, in __init__
    f"Preparing State Manager ", "Complete", "success")
  File "/usr/local/lib/python3.7/dist-packages/mudpi-0.10.0-py3.7.egg/mudpi/logger/Logger.py", line 142, in log_formatted
    return Logger.logger.log(log_level, msg)
AttributeError: 'NoneType' object has no attribute 'log'
```
I'm confused by this.  It doesn't look like the `Logger` class has an attribute called `logger`, so I guess that error message is correct that it is `None`.  Is this part of the code a work in progress or did I make a mistake somewhere during setup?

yeyeto2788, 2021-04-08:
@duffrecords in me meanwhile (and you want to, of course) you can also try the changes I have pushed to this repo https://github.com/mudpi/mudpi-core/pull/26 which should solve the issue about logging and should be up to date with latest changes done by @olixr.

olixr, 2021-05-25:
Closing with the merge of https://github.com/mudpi/mudpi-core/pull/31

closed 2021-05-25 by olixr