mudpi/mudpi-core #36: MudPi Install Error Message

deficienttech, 2021-04-25:
Had a Raspberry Pi 3 collecting dust and figured I'd try to put MudPi on it. Loaded a new SD card with a fresh install of Raspberry Pi OS. The only thing I've changed on the fresh install is the password and haven't started to lock down the Raspberry Pi yet. I followed the install guide and executed the curl command to run the bash script and ran into an issue where it seems the  can't install some core python packages - hopefully, the screenshot maybe provides some answers?

[Error Message](https://drive.google.com/file/d/1gnpZT1dmaiY0TzWKN1QhzzpFuoukoqUI/view?usp=sharing)

Not sure if this is a factor, but since the Raspberry Pi I have is the old 3 model (not the 3b+), it is running on a 32-bit OS. I ran the command **uname -m** and got the output of **armv71**, which by the power of Google, tells me that it's a 32-bit processor. Don't know if it'll matter much, but I figured I'd mention it. 

If there's a log that I can provide, let me know; tried looking at syslogs, but I didn't find anything useful there.

olixr, 2021-04-25:
Are you installing using the default install commands? 

I am asking because MudPi is in the middle of a huge update to v0.10.0. I am about to push the changes to the master branch and create the release but until then the installer tries to install the previous v0.9.1 version still. However some of the other supporting resources have changed as well so what likely happened is you tried to install and a folder structure had got changed or something. 

For the time being to install the newest version of MudPi (v0.10.0) you can use the command `curl -sL https://install.mudpi.app | bash -s -- -b feature`

The docs and all other items already support v0.10.0 as well. I just need to finish the actual merge and release then this additional command to install the other branch will not be needed. 

Please try installing again using the command I provided if you have not already. I think that will resolve your issue.

deficienttech, 2021-04-25:
That'll do it - this is what I get for thinking I can do a Sunday project :). But I was using the default install command, but I just tried your suggested command and it worked! I'll wait till you finish the merge; just ran "**$ mudpi**" after it finished and got some python errors:

[Python Errors ](https://drive.google.com/file/d/1p-AuBLCxSxKTs7XHw60nvvL16JZg3_Av/view?usp=sharing)

I imagine this won't be an issue moving forward, but I figured I'd just mention it is all. I haven't made a lot of changes to this Pi yet, so I can tear it down and bring it back fresh with no worries.

olixr, 2021-04-25:
I am glad it worked for you! I know a few people would encounter this until the update is made official. 

In terms of your next error its just looking for a config file which you have not made yet. :)

You will need to create a mudpi config file based on your components. To get started you can view the configuration docs: https://mudpi.app/docs/configuration-basics for some formatting help on basic setup. You are welcome to use any name for the file and MudPi supports both JSON and YAML. By default if you used the installer MudPi is looking for a `mudpi.config` json file in the `/home/mudpi/core/mudpi` folder. 

Have a read of the config documentation and drop a config file in that directory. You should be up an running then! 

Also as a side note I have also provided some `examples` in a folder where MudPi was installed. There are config files in there as well for reference. This is also located in the /home/mudpi/core folder.

olixr, 2021-05-25:
The main v0.10.0 changes have been merged to the master branch. I am closing this since you should not need these cross command anymore.

closed 2021-05-25 by olixr

matchico, 2022-05-11:
I am trying to install mudpi on a Pi4 4GB with raspbian and i have the same python error message. I have tried both command lines. I have updated everything, restarted, etc... No matter what the same error occurs: 
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 338, in run
    resolver.resolve(requirement_set)
  File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 102, in resolve
    self._resolve_one(requirement_set, req)
  File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 261, in _resolve_one
    check_dist_requires_python(dist)
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/packaging.py", line 55, in check_dist_requires_python
    if not check_requires_python(requires_python):
  File "/usr/lib/python3/dist-packages/pip/_internal/utils/packaging.py", line 33, in check_requires_python
    return python_version in requires_python_specifier
  File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/specifiers.py", line 676, in __contains__
    return self.contains(item)
  File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/specifiers.py", line 681, in contains
    item = parse(item)
  File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/version.py", line 28, in parse
    return Version(version)
  File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/version.py", line 219, in __init__
    match = self._regex.search(version)
TypeError: expected string or bytes-like object
MudPi Install Error: Problem installing MudPi core python package

How can i fix this?

Hacgalindo, 2022-09-15:
I have the exact same problem as above. I would really really try to get this working and try it, were you able to get past this? @matchico @olixr both command lines fail to do it. Also in manual install It gets the same problem at the point where you get to mudpi required packages "pip3 install -r /home/mudpi/core/requirements.txt"
Any thoughts?

Exception:
Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/pip/_internal/cli/base_command.py", line 143, in main
status = self.run(options, args)
File "/usr/lib/python3/dist-packages/pip/_internal/commands/install.py", line 338, in run
resolver.resolve(requirement_set)
File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 102, in resolve
self._resolve_one(requirement_set, req)
File "/usr/lib/python3/dist-packages/pip/_internal/resolve.py", line 261, in _resolve_one
check_dist_requires_python(dist)
File "/usr/lib/python3/dist-packages/pip/_internal/utils/packaging.py", line 55, in check_dist_requires_python
if not check_requires_python(requires_python):
File "/usr/lib/python3/dist-packages/pip/_internal/utils/packaging.py", line 33, in check_requires_python
return python_version in requires_python_specifier
File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/specifiers.py", line 676, in contains
return self.contains(item)
File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/specifiers.py", line 681, in contains
item = parse(item)
File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/version.py", line 28, in parse
return Version(version)
File "/usr/share/python-wheels/packaging-19.0-py2.py3-none-any.whl/packaging/version.py", line 219, in init
match = self._regex.search(version)
TypeError: expected string or bytes-like object
MudPi Install Error: Problem installing MudPi core python package