mudpi/mudpi-core #39: Install script unable to create redis.ini file in php/mods-available

agregson, 2022-05-19:
Fresh Bullseye Rapsberry pi image on Rapsberry Pi 4. Installation stops with:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: RPi.GPIO in /usr/lib/python3/dist-packages (0.7.0)
rm: cannot remove 'composer-setup.php': No such file or directory
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
redis-server is already the newest version (5:6.0.16-1+deb11u2).
The following package was automatically installed and is no longer required:
  libfuse2
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
sudo: pecl: command not found
touch: cannot touch '/etc/php//mods-available/redis.ini': No such file or directory
MudPi Install Error: Unable to create redis.ini file in php/mods-available

I have verified that /etc/php/8.1/mods-available does exist. 

Even if I touch the requested file, the install cannot continue. It seems that it cannot find the directory.

JimenezGSide, 2022-05-23:
@agregson, Im also finding this very difficult to install and use. I think ive managed to install it:
first install redis using this guide: https://amalgjose.com/2020/08/11/how-to-install-redis-in-raspberry-pi/
Then I installed php using this guide: https://pimylifeup.com/raspberry-pi-latest-php/
Then I ran the installer by ajisrael found in the issues section of github/mudpi/installer.
Sorted the config file once it was installed by watching this: https://www.youtube.com/watch?v=rAGryBYLJIg

But now when I try to go to the UI via the IP address in a web browser i get "Welcome to nginx!....further config is required...". Ive no idea how to fix this issue yet

mikedattolo, 2023-10-19:
so I figured this out, you need to install the PHP (whatever version your using) development tools because the command sudo pecl is not on the common install of php, use the command : sudo apt-get install php-pear php(version)-dev