mudpi/mudpi-core #43: Install on fresh PI fails

oliverlucas85, 2023-07-18:
**Describe the bug**
Can't install mudpi due to the following error:

Root composer.json requires php ^7.2 but your php version (8.2.8) does not satisfy that requirement.

**List any error messages / strack trace**
```
Loading composer repositories with package information
Updating dependencies
Your requirements could not be resolved to an installable set of packages.

  Problem 1
    - Root composer.json requires php ^7.2 but your php version (8.2.8) does not satisfy that requirement.

MudPi Install Error: Unable to run composer install

**To Reproduce**
Steps to reproduce the behavior:
1. Run the main installer and select Y for all except AP

**Expected behavior**
It installs successfully 

**Debug Logs**
I can't run Mudpi in debug mode as it can't install
```
Paste output log here
```

```
Paste error logs here
```

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Hardware (please complete the following information):**
 - OS: Debian GNU/Linux 11 (bullseye)
 - Board: Raspberry pi 4
 - MudPi: 0.10.0

oliverlucas85, 2023-07-18:
UPDATE:

Removed the MudPi Assistant and seems to have proceeded past the error..

Daniel434, 2025-02-17:
It appears that one of the PHP packages doesn't have a version attached to it in the install script. There is a possible resolution to this you can view in the installer repo