mudpi/mudpi-core #44: install on fresh pi results in error unable to install pip3 packages

mikedattolo, 2023-10-19:
**Describe the bug**
using a brand new Raspberry Pi 4 and flashed SD card I tried installing mudpi and as it's almost done it gets the error: unable to install pip3 packages. I have tried using commands curl -sL https://install.mudpi.app | bash and curl -sL https://install.mudpi.app | bash -s -- -b feature and avail

Paste the error message here: 
Unable to install pip3 packages

**To Reproduce**
Steps to reproduce the behavior:
type curl -sL https://install.mudpi.app | bash into terminal and wait

**Expected behavior**
should install Mudpi no probelm

**Hardware (please complete the following information):**
 - OS: [e.g. Raspberry OS newest 64bit]
 - Board [e.g. raspberry pi 4,]
 - MudPi Version [e.g. v0.10.0]

mikedattolo, 2023-10-19:
ok so I now fixed my problem with the error however now when installing I get the error: problem installing composer

wfdonker, 2025-02-12:
what is the solution for pip 3, I get the same message.